Hyesu/capstone_carcar5talk #5: Pictures

Hyesu, 2015-03-10:
사진 올라갈 전용티켓입니다. 
[image]

parkhoon, 2015-03-20:
[image]

Hyesu, 2015-03-23:
닛산 리어뷰 미러 비교 사진
[image]

[image]

Hyesu, 2015-03-23:
usecase diagram

[image]

[image]

SungjungKim, 2015-03-23:
[image]

Hyesu, 2015-04-01:
[image]

Hyesu, 2015-04-01:
[image]

Hyesu, 2015-04-01:
[image]

Hyesu, 2015-04-01:
[image]

SungjungKim, 2015-04-01:
/home/sungjung/다운로드/Bluetoot_operation.jpg

Hyesu, 2015-04-07:
[image]

[image]

SungjungKim, 2015-04-07:
[image]

SungjungKim, 2015-04-07:
[image]

SungjungKim, 2015-04-15:
[image]
[image]
[image]
[image]

SungjungKim, 2015-05-24:
[image]