Hyesu/capstone_carcar5talk #18: Implement Ad-hoc Communication between Raspberry Pis

Hyesu, 2015-03-12:
Due Date: 2015-04-08
parkhoon with tnm1264

Hyesu, 2015-04-01:
[라즈베리파이 키보드 설정-더이상 따옴표와 파이프때문에 고생하지 마세요~!](http://www.rasplay.org/?p=3786)

Hyesu, 2015-04-07:
우분투 라즈베리파이에 깔고, "read-only file system"과 같은 에러 메세지 나오면서 파일 생성 및 수정이 불가능한 경우!!  
```
#mount -o remount,rw /  
```  

입력하세용

parkhoon, 2015-04-07:
## Ad hoc mode Setting on SNAPPY UBUNTU CORE

before the edit configure file , type this command first

> sudo mount -o remount,rw /

after mount command then you can modify a interfaces file where is,

> /etc/network/interfaces

You can use any editor like vi, vim, nano ,, 
I used vi editor

> sudo vi /etc/network/interfaces

you may see below codes .. 
```
# interfaces(5) file  used by ifup(8) and ifdown(8)
# include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
```

and edit this codes like this

```
1: auto lo
2: 
3: iface lo inet loopback
4: iface eth0 inet dhcp
5: 
6: auto wlan0
7: allow-hotplug wlan0
8: iface wlan0 inet dhcp
9:         address 192.168.42.1
10:        netmask 255.255.255.0
11:        wireless-essid PiAdHoc
12:        wireless-channel 6
13:        wireless-mode ad-hoc
```

actually we need only 6~13 lines 

after edit, save & quit

to apply this configure you need to ifdown & ifup command or reboot

> sudo ifdown wlan0
> sudo ifup wlan0

or 

> sudo reboot

if you configure PI1 like this , then you need to make ad-hoc mode another pi with different address
for example, 
> PI1 :  address 192.168.42.1
> PI2 :  address 192.168.42.2
> ...

after these setting check you interfaces information

>> ifconfig
>> iwconfig

Pis are must in same cell , if not, your setting has error somewhere.

If you have no error , Pis are can ping to each other.

Hyesu, 2015-04-09:
중간보고서 Ad-hoc 부분 위키에 정리해주세욤~

Hyesu, 2015-05-12:
Due Date: 05.14

tnm1264, 2015-05-23:
Ad-hoc 방식은 2학기 Capstone 에서 다시 구현해 볼 예정이고, 5/28 전시회까지는 DHCP 방식으로 구현할 예정입니다.

closed 2015-05-23 by tnm1264

lwsang13, 2015-08-05:
안녕하세요...제가 요즘 고민하고 있는 부분과 흡사한 부분에 대해서 구현하고 계신거 같은데 질문 하나만 받아 주시면 안될까요...?
혹시 raspberry pi에서 ad hoc 구현할 때 wifi 모듈을 어떤 모델 사용하시는지가 궁금합니다. 제가 사용하는 모듈은 아무리 설정값을 조정ㅇ해서 ad hoc에서 ping이 안가더라구요...

감사합니닷!

tnm1264, 2015-08-05:
에디맥스를 쓰시는게 가능성 있을겁니다.realtek rtl8188cus 모델은 pi에 드라이버가 안깔려요....저희도 이문제 때문에 1달 넘게 고생하다가 시간없어서 DHCP 모드로 바꿨습니다..커널 컴파일을 해도 노트북끼리는 adhoc이 되는데 pi끼리는 안되더라구요ㅠㅜ

lwsang13, 2015-08-05:
ㅠ_ㅠ 답장 감사드립니다!!
저도 realtek rtl8188cus 모델로 며칠 고생하다가 RTL8192CU 칩셋 사용하는거로 새로 구매해서 해볼려고 하는데 이것도 안될까요?
그리고 DHCP 모드라는게... Ad-hoc처럼 내부망 만들어서 스스로 라우팅하고 이런게 가능한 건가요?

tnm1264, 2015-08-05:
Pi하나가 DHCP서버가 되어서 다른 Pi에서 ip할당 해줘요. 외부랑 인터넷 연결없이! 제 생각엔 Pi에 해당 드라이버를 설치할 수만 있으면 8188 이든 8192 든 다 될겁니다. 저희도 8192로 새로 샀는데 Realtek홈페이지 가보면 8188부터 8192까지 같은 드라이버에요. 드라이버 설치 꼭 성공하길 빕니다.