Hyesu/capstone_carcar5talk #8: Implement Bluetooth Pairing Module(Raspberry Pi)

Hyesu, 2015-03-12:
Due Date: 2015-03-18

SungjungKim, 2015-03-19:
sudo apt-get update
    sudo apt-get install bluez bluez-utils minicom

SungjungKim, 2015-03-19:
Bluez PNAT is not needed and breaks things, so edit main.conf:

sudo nano /etc/bluetooth/main.conf

SungjungKim, 2015-03-19:
Add this line towards the top:
    DisablePlugins = pnat

SungjungKim, 2015-03-19:
You'll need to reboot after this; restarting bluetooth doesn't seem to work.
Now set the Pi to be discoverable, and pair it with your mobile device:
    sudo hciconfig hci0 piscan
    bluez-simple-agent

SungjungKim, 2015-03-19:
At this point I used the phone to find the Pi and do the pairing. This is different depending on your bluetooth dongle, so watch the Pi and follow the prompts. Control-C when the pairing is succesful. This needs be done just once.
Now we want to add the Serial Port Protocol on the Pi, and start listening for connections.
    sdptool add sp
    sudo rfcomm listen hci0

SungjungKim, 2015-03-19:
This should respond with "Waiting for connection on channel 1"
Back on your mobile device, open a bluetooth terminal program. On Android I found "Bluetooth Terminal" from QWERTY works well and is less ugly than others I tried. Choose "Connect a device - insecure" and it should connect. Your Pi terminal should say "Connection from AA:BB:CC:DD:EE:FF to /dev/rfcomm0". These steps need to be done each time the Pi is restarted; you can automate them be putting the following into your /etc/rc.local file:

    printf "Adding SPP and listening on rfcomm0\n"
    sdptool add sp; rfcomm listen hci0 > /dev/null 2>&1 &

SungjungKim, 2015-03-19:
However, I found that I still need to login to the Pi in order for it to accept connections. I don't know why.
To test, send the session into the background with Ctrl-z, or open a new terminal, and start minicom:

    minicom -D /dev/rfcomm0

SungjungKim, 2015-03-19:
Type "Hello Pi" and press send on the phone, and it should show in minicom. Type "Hello phone<enter>" in minicom and it should show up on the phone. If you're not familiar with minicom, note that you won't see what you're typing, and it won't be sent to the phone until you hit enter. Use ctrl-a x to exit minicom.

I haven't tried serial programming on the Pi, but something like PySerial should make it easy.

Hyesu, 2015-04-01:
https://github.com/Hyesu/capstone_carcar5talk/issues/23  
요거랑 같이 위키에 정리하고 이슈는 닫아주세영~

closed 2015-04-09 by Hyesu