Hyesu/capstone_carcar5talk #54: Installation Manual 작성

Hyesu, 2015-05-12:
각자! 아무것도 없는 파이에 깔아야 하는 것들 작성.
OS, module, driver, library 등등

Due Date: 05.26

Hyesu, 2015-05-21:
* python IPC를 이용하기 위해 posix_ipc라는 패키지를 다운 받아야 함.  
posix_ipc 패키지는 capstone_carcar5talk/에 있고, tar 압축 해제한 뒤 해당 폴더에서 다음 명령 실행  
```
$ sudo apt-get python-dev
$ python setup.py install
```

Hyesu, 2015-05-22:
* wiringPi를 이용하기 위해 wiringPi 라이브러리를 다운받아야 함.
```
$ git clone git://git.drogon.net/wiringPi
$ cd wiringPi
$ git pull origin
$ ./build
```

Hyesu, 2015-05-22:
* Network_Client_Send.py를 실행하기 위해서 python wifi module 설치를 해야 함.

parkhoon, 2015-05-23:
gps 모듈의 uart 통신을 하기 위해서 다음과 같은 설정을 해야 한다.
```
$ sudo vi /boot/cmdline.txt
# "console=ttyAMA0" 부분을 삭제

$ sudo vi /etc/inittab
# ttyAMA0 줄 주석 처리
```

SungjungKim, 2015-05-25:
각 차량 정보는 블루투스 통신을 이용하여 라즈베리파이에서 안드로이드로 전송된다.
라즈베리파이에서 블루투스 통신을 이용하기 위해 다음과 같은 설정이 필요하다.
```
$ sudo apt-get update
$ sudo apt-get install bluez python-bluez
$ sudo apt-get install bluetooth bluez-utils blueman
$ sudo vi /etc/bluetooth/rfcomm.conf
{ add this property }
rfcomm1 {
   bind yes;
   device xx:xx:xx:xx:xx:xx;
   channel 1;
   comment "Connection to Bluetooth module";
}
$ sudo rfcomm bind all
$ sudo hciconfig hci0 up
$ sudo vi /etc/group
{ change line 8 from “lp: x:7:”  to  “lp: x:7:pi” save and exit }
$ sudo vi /etc/bluetooth/main.conf 
{ add line DisablePlugins = pnat }
$ sudo reboot
```

위의 환경 설정을 마쳤다면, 안드로이드에 라즈베리파이를 오토페어링 시켜놓는 작업이 필요하다.
라즈베리파이에서 다음과 같은 명령어를 실행하여 라즈베리파이 블루투스를 안드로이드에서 스캔 가능하도록 설정해준다.
```
$ sudo hciconfig hci0 piscan
```

그 다음 안드로이드에서 블루투스 스캔 모드를 통해 라즈베리파이를 페어링시킨다.